Duet3D/PanelDueFirmware #299: Display thumbnail on Status page during prinitng

dc42, 2023-05-02:
### Description

See https://forum.duet3d.com/post/314263 for suggested layout; Questions:

- will the standard 160x160 thumbnail fit in that space?
- what would be do if there were more tools/heaters filling that space?

mfs12, 2023-10-31:
Hey @dc42, the space is pretty limited and tools and heaters can use up the complete space.

MarkoP2507, 2024-10-19:
I would also like to have a preview

You could make an option in the settings whether you want the preview or other tools

dc42, 2025-05-09:
We don't plan to do this. PanelDue is being superseded by a new screen I will continue to implement changes where they are needed to retain compatibility with new RepRapFirmware versions for some time, but it is unlikely that we will implement new features.

closed 2025-05-09 by dc42