Duet3D/PanelDueFirmware #308: Report no files found if gcode folder is empty

mfs12, 2023-08-10:
### Description

## Description

From slack:
> With help from I have discovered after much frustration that if you try to test a PanelDue's ability to read SD card 0, the one on he uet, when the Gcodes directory is empty (as it is on a new Duet with a factory-flashed SD card image), instead of reporting that the directory is empty, after a long wait it reports "Error 2 accessing SD card". This is not documented anywhere that I can find, and I think that it should be - it would have saved me quite a few wasted hours. Even better if the PanelDue just reported "No files found" or similar.

> can you log this as a bug and look to implement a fix. Should be straightforward to reproduce.of you have no files or subdirectories in the /gcodes directory

## Versions
- paneldue 3.4.1
- rrf 3.4.1

## Process
- of you have no files or subdirectories in the /gcodes directory

## Experienced Result
- error 2 is reported after long time

## Expected Result
- no files found is reported

T3P3, 2023-08-30:
No Files Found

mfs12, 2023-08-30:
> No Files Found: SD Card 0
> 
> or if this is SD card 1:
> 
> No Files Found: SD Card 1

but the info about the sd card is redundant. At the top it's already displayed which SD card is currently displayed. Do you still want it?

T3P3, 2023-08-30:
ah yes fair enough, just "No files Found" then

PR #310 (2023-08-30): report no files found in file popup
Closes #308.
closed 2023-08-30 by mfs12