Duet3D/PanelDueFirmware #317: Two digits for the display of x and y coordinates

wschadow, 2023-10-21:
### Description

The x and y coordinated are displayed with one digit:

[image]

Especially for the calibration of an IDEX machine it would be very helpful if two digits would be shown.

mfs12, 2023-10-31:
Hey @wschadow, do you request those two digits for DuetWebControl or PanelDue? This is a bit unclear.

If PanelDue please share the right screenshot, if DuetWebControl then this issue needs to be moved to DuetWebControl project.

wschadow, 2023-11-04:
Sorry for the confusion, I meant both but submitted the wrong screenshot. I just created a pull request for the changes in the PanelDue firmware. 

Thanks for fixing the display of the buttons already!

mfs12, 2023-11-27:
Needs to be reverted as support for values up to 999 are necessary.

PR #324 (2023-11-10): 317 two digits for the display of x and y coordinates
Closes #317.
closed 2023-11-28 by mfs12