Duet3D/PanelDueFirmware #335: Menue not displayed properly

wschadow, 2024-05-21:
### Description

Before posting a bug report here, please check the forum and gitlab issue tracker if it was already reported.

- https://forum.duet3d.com/topic/35223/menue-not-displayed-properly

## Description
Depending on the number of items sometimes there are empty bars displayed.

## Versions
3.5.0, 3.5.0-rc9, 3.5.0-rc8

## Process
Please add a step by step description on how to reproduce the bug.

## Experienced Result
[image]

## Expected Result
Please describe the expected result.

dc42, 2024-12-09:
@wschadow please describe what you think is wrong with this display. If you are looking to order the macros on the screen, see https://docs.duet3d.com/en/User_manual/Tuning/Macros_tasks#naming-and-ordering-macros for how to do that.

wschadow, 2025-05-06:
There are empty fields displayed

[image]

dc42, 2025-05-09:
@wschadow please provide a directory listing of your /macros folder. I am working on a new release and I would like to fix this if I can confirm that it is a bug.

wschadow, 2025-05-13:
[macros.zip](https://github.com/user-attachments/files/20191800/macros.zip)

[image]

The screenshot is from the directory macros/03-Diagnostics

dc42, 2025-05-13:
Thanks. However, in the macros.zip file you posted there is no folder called macros/03-Diagnostics:

[image]

Perhaps the SD card has been corrupted? Please move the SD card to a PC or laptop, then run a disk checking tool on it. If your PC runs Windows then to do this you can right-click on the corresponding drive letter in Windows Explorer, select Tools, then Error Checking.

wschadow, 2025-05-14:
please see under macros/04-Maintenance/03-Diagnostics

dc42, 2025-05-20:
Thans. Unfortunately I can't reproduce that issue. I renamed the /macros folder on one of my SD cards and extracted /macros from your zip file on to the SD card. I don't see any spurious (blank) entries on PanelDue.

It could be that your SD card has some hidden entries that didn't make it into the zip file. Please help me investigate further as follows:

1. On a given machine, do you always see the same blank entries each time you visit that subfolder of macros? Or does it depend on what you have previously viewed on the PanelDue?
2. If you power the machine (including the PanelDue) off and on again, then as soon as PanelDue has connected and finished displaying messages you go straight to that macros folder on PanelDue, is the blank entry there already?
3. Please run the following command from DWC and report the response (this is the command that PanelDue uses to fetch the file list): `m20 s2 P"/macros/04-Maintenance/03-Diagnostics"`

dc42, 2025-05-21:
Thanks, that suggests a regression in PanelDueFirmware between 3.4 and 3.5. I'll look at the changes.

dc42, 2025-05-21:
@wschadow there were a lot of changes between 3.4.0 and 3.5.0. As I am unable to reproduce this issue, whereas you appear to be able to reproduce it easily, would you mind narrowing down which pair of releases the bug was introduced between? You can find all the releases at https://github.com/Duet3D/PanelDueFirmware/releases. I suggest you start by trying a release about midway between 3.4.0 and 3.5.0 such as 3.5.0-rc.5 and see whether the bug is present in that one. Then continue to binary chop the release range until you get down to a pair of adjacent releases which do and do not exhibit the issue.

wschadow, 2025-05-22:
The bug is introduced from rc8 (which is ok) to rc9. Just for info there was another issue with rc7 that displayed too many items and was resolved in rc8 but is maybe related https://github.com/Duet3D/PanelDueFirmware/issues/316

dc42, 2025-05-27:
Thanks, that should help me to narrow it down.

wschadow, 2025-05-28:
We have 5i and 7i. I just tried your latest 3.5.2-alpha version and empty fields are gone :)

dc42, 2025-05-29:
Thanks for the feedback! I wasn't able to definitively identify the cause of the issue you reported, but I established that the change between rc8 and rc9 invoked "undefined behaviour" according to the C+ standard. I corrected the code to avoid that in the 3.5.2 alpha version. I will release that correction, along with another change in that alpha build that is needed to support the RRF 3.7 firmware series).